Fmstrat/diy-ipmi #2: Alternative video capture system

tpisto, 2018-01-07:
Hi! I can confirm that with this I was able to view the screen:

https://www.ebay.com/itm/Mini-Portable-1080P-USB2-0-HD-HDMI-60fps-Video-VGA-Capture-Monitor-Card-for-PC/122769635585

It simplifies the screen route since you can just plug it to your RPI USB port and connect other end to the server HDMI port. It also does not need external power (at least when I tested with only one attached)

Fmstrat, 2018-01-07:
Awesome. Can you show me the output of `lspci -a` and `lsusb`? And any model numbers on the device? I'd like to see if I can find a standard retailer vs eBay so the link stays active. Thanks!

tpisto, 2018-01-07:
There is no model names or numbers in the device. The lsusb shows: Bus 001 Device 005: ID 1b71:3002 Fushicai USBTV007 Video Grabber [EasyCAP]

https://www.linuxtv.org/wiki/index.php/Easycap#USBTV007_EasyCAP

Fmstrat, 2018-01-17:
OK, thanks. I imagine the issue is that the resolution of the monitor visual doesn't line up. So we need a configuration for the output resolution of the device. Would that work?

sammcj, 2018-01-17:
FYI - This is the card I'll be using, and what @ross-w is already using in his own trials of implementing a pi-kvm.

Fmstrat, 2018-03-23:
I'm going to pick a few of these up and get them implemeted into this project at some point in the near future.

SterlingButters, 2018-04-07:
I was just about to suggest this as well

seffyroff, 2018-05-10:
This is exactly the iteration I was looking for to pull the trigger on building my own, thanks for the legwork guys!

seffyroff, 2018-06-06:
My dongle arrived today, and it works pretty good so far.  I am wondering about integrating something like  OpenIPMI so some interesting tools become integratable, but that's on the horizon :)

Now to reboot some servers, whee!

holmesb, 2019-02-17:
Anyone get very blurry display using this [Fushicai USBTV00](https://www.ebay.com/itm/Portable-HD-USB-2-0-HDMI-1080P-60fps-Monitor-Video-Capture-Card-Adapter-For-PC-/152733483118?oid=122769635585)?  :
[image]
Can't read the text.  Anyone fix this?

eddie4, 2019-05-02:
I was looking into building something like this project myself including a 4G dongle with openvpn.

During my search I came across
https://nl.aliexpress.com/item/Kebidumei-USB-3-0-Game-Video-Capture-Box-USB3-0-1080-P-60FPS-HDMI-Live-Streaming/32860650057.html 

Haven't bought it yet, but thought i would drop it here.

sammcj, 2019-05-02:
@eddie4 - That will only do video output though right?

eddie4, 2019-05-03:
@sammcj  - Well no it's a capture device "You can use any 3rd party software to capture or stream your HD game or other HD video source."

The output is so you can connect a monitor as well as record.

dmk1000, 2019-06-06:
@holmesb 
The UTV007 is an HDMI to analog converter and then analog to USB converter in one package:
http://www.epanorama.net/newepa/2018/06/19/cheap-poor-hdmi-to-usb-converter/

The bluriness is due to going via analog and wrong channel selection (s-video/composite). I improved the picture on mine by changing image.php to add "-channel 0" before the -i. It's never going to be good though.
ie
`system("avconv -f video4linux2 -channel 0 -i ". $_GET['vid'] ." -vframes 1 -s 720x480 -v quiet -y ". $filename);`

[image]

dmk1000, 2019-06-06:
@eddie4 
I've bought one of those and I've only got it to work at all on OBS on Windows 10 where it shows up as a "USB Video Device" under Cameras in Device Manager and then only on some USB ports. The picture does look good though so I'm trying to persevere with it.

Under Linux (raspbian/Linux raspberrypi 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux) I get the following repeated over and over in dmesg:
```
[261216.959505] usb 1-1.5: new high-speed USB device number 118 using dwc_otg
[261217.096992] usb 1-1.5: New USB device found, idVendor=1e4e, idProduct=7103, bcdDevice= 1.00
[261217.097014] usb 1-1.5: New USB device strings: Mfr=6, Product=7, SerialNumber=3
[261217.097027] usb 1-1.5: Product: USB Video Device
[261217.097038] usb 1-1.5: Manufacturer: USB Video Device
[261217.097049] usb 1-1.5: SerialNumber: 20000130041415
[261217.099008] uvcvideo: Found UVC 1.00 device USB Video Device  (1e4e:7103)
[261217.111644] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
[261217.112743] uvcvideo 1-1.5:1.0: Entity type for entity Extension 4 was not initialized!
[261217.112766] uvcvideo 1-1.5:1.0: Entity type for entity Processing 3 was not initialized!
[261217.112780] uvcvideo 1-1.5:1.0: Entity type for entity Camera 1 was not initialized!
[261220.501854] usb 1-1.5: USB disconnect, device number 118
```
The HDMI out looks fine on my TV into it's HDMI port but when I try to use an oldish LG monitor with DVI input (HDMI to DVI cable) the monitor fails to display anything cycles through detecting and not detecting a signal.

eddie4, 2019-06-07:
That quality is pretty good whats the frame rate like? If you get stuck I wouldn't mind taking a look over SSH.

dmk1000, 2019-06-07:
Hi @eddie4,
The frame rate in Window 10 (Ryzen 1600) via OBS over USB3 is fine. I don't know the exact number but the navigating about the BIOS screen looked smooth enough in OBS. How that would work with the image.php script if it ever works on the Pi, I don't know :)

I couldn't get it to display an image at all on VLC on the Windows 10 PC but nor did VLC give an error.
I was trying it on the Win 10 machine to see it it did actually do anything before deciding if I should send it back. Front case panel USB 3 ports didn't work, but rear motherboard USB 3 sockets did.

The Raspberry Pi is a 3(something) with the latest Raspbian and firmware update.

I do have another AMD PC I can install Win 10 and Linux on to see if I can get it to work first under Windows 10, and then at all under Linux using the same USB socket.

I don't have a mains powered USB hub, and it looks like they aren't easy to come by in brick & mortar stores so it'll be next week before I can get one delivered.

I know nothing about UVC. I assume that things are just supposed to work based on info provided by the device and the fact that USB ID 1e4e:7103 doesn't appear in any searches shouldn't necessarily mean it won't work without driver work.

dieserkai, 2019-06-28:
have you guys seen this: https://auvidea.eu/b101-hdmi-to-csi-2-bridge-15-pin-fpc/ it looks quite promising.

eddie4, 2019-07-02:
@dieserkai looks interesting but the "HDMI input: 1080p25 only (software limitation at this time)" would be a problem.

f0ff886f, 2019-08-09:
@eddie4 Why is 1080p25 so bad? for basic terminal/bios/etc 25fps is fine, 1080p is amazing.

closed 2024-12-20 by tpisto