Fmstrat/diy-ipmi #3: Add mount virtual CD / ISO & USB img option

sammcj, 2018-01-17:
Would be _very_ useful for mounting recovery ISO's, reinstalling an OS, applying firmware updates etc...

Fmstrat, 2018-01-17:
I'm not sure what you're asking here. I end up keeping a USB thumb drive in the server all of the time. When booted, I can then copy BIOS updates to it, then use the IPMI to reboot and update. Would this not work for you?

sammcj, 2018-01-17:
There’s actually no need for the full raspberry pi as well as the zero, due to the zero being able operate one of its USB ports in OTG (bidirectional) mode and the ability to bridge connections (aka hit a reset or power switch) across pins.

Just like normal IPMI, ILO, ILOM, DRAC etc interfaces there is the opportunity to mount an iso/img from your remote machine via the web interface, this functionality is all available within the kernel.

See https://gist.github.com/gbaman/50b6cca61dd1c3f88f41 as an example.

Either myself or a colleague may have a MR for this functionality in days to come.

sammcj, 2018-01-17:
Also note: not asking you to do the work for something you don’t need, but it’s something myself and a few others do / will require so, we’re either planning modifying and updating your project with an MR or rolling our own IP-KVM / IPMI project with a single pi-zero, figured it’d be best to first consider working as a community but if it’s outside the scope of what you’re interested in - that’s ok :)

sammcj, 2018-01-17:
Somewhat related but out of scope, see: https://smcleod.net/tech/update-rcd-stonith-design/ I designed this for issuing a reset or power control to a host in a complex cluster, I believe the pi-zero can perform similar bridging of contacts across reset or similar pins ;)

Fmstrat, 2018-01-17:
@sammcj Very, very open to collaboration, and to simplifying the project. And would love the integration of remote USB mount, I just wasn't following the request. 👍 A few thoughts as you and your colleague begin contributions:

- I've created a `dev` branch for PRs, I think this branch will be great for simple changes you may find along the way, such as what you caught with PHP7, or things like remote mount USB.
- I've created a `zero` branch for the big shift of using only a Pi0 as this will likely be a major change in the process. 
- For resetting using the Pi0, my only ask is to ensure there is no increased risk of electrical faults on the outputs to the PC. That was one of the reasons for using the relay board vs pins as the selected board has magnetic and optical isoloation.

I'd ask that any changes that will work in both setups be completed in the `dev` branch first, that way it's easier to merge and if other contributors come in, they (and myself) might be able to follow easier. However, you will likely be more of a contributor than myself at this point. If that starts to happen, I'll just make you an admin on the repository.

sammcj, 2018-01-17:
Great, thanks @Fmstrat, I can't promise I'll be doing within the next few days as work is hectic, but it's great you're all for community and thanks for inviting me, my colleague who was working on a similar idea, but will now likely submit a few MRs to this project is @ross-w 👍

Turbine1991, 2018-03-31:
@sammcj Heck, one can also use a rather affordable arduino for keyboard/mouse. https://www.arduino.cc/en/Reference/Libraries

Fmstrat, 2018-03-31:
@turbine1991 I thought the Pi0 and Ardiuno were comparable in price?

microbug, 2018-04-11:
@sammcj @Fmstrat one problem with just using a Pi Zero would be that it doesn't have an integrated ethernet port. If the USB OTG port is in use (presumably with `g_multi` presenting ethernet, serial and HID interfaces to the host) and the host goes offline, the Zero is stuck. Unless you were thinking about the Zero W that comes with WiFi, though I wouldn't personally want to risk my IPMI with a potentially flakey WiFi connection inside a metal rack cabinet.

If you were thinking of using a USB hub this won't work, as in order to emulate an HID keyboard the Pi would have to be able to use USB OTG, and this does not work with USB hubs (this is why the proper Pi 3(+) can't do HID emulation through USB OTG, as it has a USB hub on-board to provide its multiple USB ports).

As far as I can see you'd have two options:

- Use a solution like the ENC28J60 that can run off a Pi's GPIO. This only provides 10BASE-T (10Mbps) so possibly a bit touch-and-go given that total throughput won't be that high and the Pi needs to push video through it.
- Use a different single board computer

As I mentioned in #6 I'm going with the Orange Pi Zero, which has 100BASE-T ethernet and a USB OTG port. I'll let you know how it turns out, though it might be a while as all the stuff has to come from China.

closed 2022-11-13 by sammcj