Fmstrat/diy-ipmi #6: Possible Alternative SBC to lower cost

JFish222, 2018-03-23:
Thanks for creating this project! I haven't dipped my toe in the water yet (excited that the rpi B+ is out!)

I noticed that there are two SBC's that feature HDMI-in and cost ~$100:

RockChip Based (not sure if supports OTG):
http://www.orangepi.org/Orange%20Pi%20RK3399/

Qualcom Based:
https://www.inforcecomputing.com/products/application-ready-platforms/qualcomm-snapdragon-600-inforce-6420t

The questions:

1.  Given that these handle encoding, would these present a simpler platform?

1.  Would USB OTG solve the need for a USB-to-Serial adapter + Pi0?  (If I understand what's happening here, the RPi0 is emulating a USB device?

1.  Any thoughts on implementing? (I'd pitch in, but I barely know su from sudo ... linux development is a bit beyond me.)

JFish222, 2018-03-23:
So!  Just to not be lazy I've continued my Google Search of OTG ports emulating keyboard and mouse.

I now understand why this couldn't be done with the RPi's OTG port [due to the included USB Hub](https://raspberrypi.stackexchange.com/a/32199).
And I caught the adafruit article on emulating keyboard mouse with the [RPi0].
 I'm guessing the inspiration for your solution?  (https://learn.adafruit.com/turning-your-raspberry-pi-zero-into-a-usb-gadget?view=all)

It looks like the OTG port solution[ will actually work provided the hardware will support it](https://bbs.nextthing.co/t/using-usb-otg-to-act-like-a-keyboard/769/7)!

Fmstrat, 2018-03-23:
One of those devices may work, but I probably won't make the investment to reinvent something I have working already ;)

You could lower cost two ways:
- Use a Teensy instead of a Pi0 (minimal savings)
- Use the HDMI capture devive mentioned here: https://github.com/Fmstrat/diy-ipmi/issues/2

The other rationale is that the Pi3 can administer multiple computers by adding multiple capture and control devices. The scalability keeps things more affordable if you have more than one machine since you only need one main board. Buying multiple $99 boards would get pricey.

Turbine1991, 2018-03-31:
JFish22 The orange pi board is compatible with usb otg. The type c specification has it built in. Hence why the prior boards had a dedicated port started OTG in place of where the type c is located.

Turbine1991, 2018-04-01:
A word of warning for both the Inforce 6420 &  Orange Pi RK3399. They require a 5v 4a power input and 12v 2a. This requires an external and specific power supply.

A typical pc usb plug would supply anywhere between 500ma-2.1a. Furthermore, everyone's setups are different. So even using a Y cable for power wouldn't be a decent solution.

I think the best idea (besides svideo), is using a HDMI to CSI-2 bridging chip. The board by Auvidea is a bit of the pricy side with incredibly high set shipping fees. I could probably create one of those bridging boards, I'll just need to acquire a better heatgun.

JFish222, 2018-04-02:
Hi @Fmstrat, 
Completely understand!  This will definitely not scale for a server room environment.  

In my case, I was looking for a cost-effective and simple way to support remote PC's (ex: parents out of state) in a pre-boot environment.  Similar to a Lantronix Spider, a field tool for single instance remote support. (My parents out of state just ran into corrupted win 10 environment and TeamViewer and the like were a no-go.)

Hi @Turbine1991,
Thanks for the word of warning!  I have a 5v 2.4a for my Pine64 board and didn't realize that power draw was getting that high!

Wow, those Avudida chips cost as much as the Inforce board and they both use Toshiba encoders (which sell for $6 in lots of 1.5k)!  That's a healthy markup.

microbug, 2018-04-09:
I had a look for an SBC that could handle everything but wasn't unavailable/overpowered/expensive like the RK3399-based example above. So far I've come up with the Orange Pi Zero, which has a USB OTG port that can be used for gadgets, and 2 USB 2.0 host ports available via the header (easily accessible if you get the additional development board for a little extra). Combining the USB capture device from #2 and an Orange Pi Zero + development board should simplify the solution significantly.

I'm ordering the parts mentioned and (time permitting) will let you know how it turns out.

DedupOperator, 2018-09-21:
I'm very excited to find out about this project
Ever since 5 years ago when Auvidea founder kickstarted the first board, RPI GPU was still fully closed source and I understood it will take some time to get it done to market.
Times passed so I did some research about SBC's with HDMI input and here are my findings:

Banana Pi BPI-W2 (RTD1296 chip with hdmi input):
-----------------------------------------------------------------------
http://wiki.banana-pi.org/Banana_Pi_BPI-W2
https://bananapi.gitbooks.io/banana-pi-bpi-w2-with-realtec-rtd1296/content/
https://forum.armbian.com/topic/7376-banana-pi-bpi-w2/
https://github.com/BPI-SINOVOIP/BPI-W2-bsp

purchase links:
https://www.aliexpress.com/store/product/Banana-Pi-BPI-W2-smart-NAS-router-RTD1296-chip-design/302756_32862157238.html
https://item.taobao.com/item.htm?id=567700943779

kits:
https://www.aliexpress.com/store/product/Banana-pi-BPI-W2-smart-router-with-Realtec-RTD1296-Design-Suitable-for-Home-Entertainment-Home-automation/1382212_32903027440.html

Orange Pi R2 (future?) SBC:
---------------------------------------
https://www.cnx-software.com/2018/04/12/banana-pi-bpi-w2-multimedia-router-nas-board-launched-for-93/#comment-553063

1) I think the best method for this project will be using digital signal so capturing recording could be compressed to minimum and could also help manipulation and automation.
The mentioned SBC and Auvidea boards should fit those requirements hopefully.

HDMI input preffered.
2) VGA should also be used as servers use it as default display-out.

Any thoughts about VGA input? It should have much better signal than the cheap composite USB capture devices.
I've got the [Epiphan DVI2USB](https://www.epiphan.com/wp-content/uploads/2015/06/Videograbbers-EOL-Transition.pdf)
I bought it some time ago cheaply on ebay for trying to hacking with a similar project, I could test it for this purpose.

DedupOperator, 2018-10-12:
some news:
1) Linux kernel source for the Banana PI BPI-W2 (with RTD1296 chipset) released:
https://github.com/BPI-SINOVOIP/BPI-W2-bsp

2) according to the comment in the link provided, Orange PI R2 will be released in a few weeks:
[cnxsoft comment](https://www.cnx-software.com/2018/09/23/realtek-rtd1296-u-boot-linux-source-code-rtd1619-cortex-a55-soc/#comment-556299)