Fmstrat/diy-ipmi #8: how to wire a pc power switch to the relay card

timoyuen, 2018-04-14:
thank you for the wonderful project.

i am not a hardware person.

might i know how to wire the power switch to the relay card? thanks again!

SterlingButters, 2018-04-19:
I also noticed this was missing from tutorial

DedupOperator, 2018-09-21:
In your case, take a look at this:
[SwitchBot](https://www.switch-bot.com/)
[Bluetooth Control by RaspberryPI](https://github.com/OpenWonderLabs/python-host)

nonfatalexec, 2018-10-12:
I'm also not a hardware person and waiting for components to arrive. From reading [http://www.tomshardware.com/answers/id-3351373/wiring-parallel-power-reset-button-diagram.html](http://www.tomshardware.com/answers/id-3351373/wiring-parallel-power-reset-button-diagram.html) and a manual from my motherboard, I got the following:
1. Pushing the power/reset button closes the connection between 2 pins that are behind the button. This kind of switch is "normally open".
2. The relay card has pins that are named (NO1, COM1, NC1), (NO2, COM2, NC2), etc. To control the **power** button, a "normally open" pin (eg. NO1) and a COM pin (eg. COM1) needs to be connected to the 2 pins behind the **power** button.
3. Putting power to this wiring would close the connection to the 2 pins behind the **power** button to do the same thing as pressing the **power** button.

spyd3rweb, 2018-11-10:
Attached are images of the custom y-splitter cables I made for my eight Dell T1700 (custom jst connector).  Most folks with standard motherboards should be able to use cheap 4-conductor wire with regular dupont connectors to make the y splitter.
(y-crimp: https://www.youtube.com/watch?v=8W-zdo1AVns) 
(y-solder: https://www.youtube.com/watch?v=ZeiS8tdCbLU)
I recommend adding in another set of wires for the power leds, just in case diy-ipmi ever implements the ability to monitor current power state (https://github.com/Fmstrat/diy-ipmi/issues/30):
[image]
[image]
[image]

nonfatalexec, 2018-11-15:
> Attached are images of the custom y-splitter cables I made for my eight Dell T1700 (custom jst connector). Most folks with standard motherboards should be able to use cheap 4-conductor wire with [...]

How are the y-splitters used for connecting the relay card to the PC power/reset switch?

spyd3rweb, 2018-11-18:
@nonfatalexec 
The point of the y-splitter is to allow the use of both the relay and the physical power button on the PC to act as the power/reset switch.  The power/reset switch pins of the Y-splitter (from the motherboard) are connected to the common and normally closed terminals of the relay.  I placed the line/load-power wire (providing 3.3V) to the common terminal and the load wire to the normally closed terminal.  
# Useful references/instructions for relay output wiring:
- (http://s3.amazonaws.com/s3.image.smart/download/101-70-C01/101-70-103.jpg)
- (https://www.youtube.com/watch?v=5NxVmg8ZFEc#t=3m21s)
[image]
[image]