Fmstrat/diy-ipmi #9: Can't access Pi0 via Serial

SterlingButters, 2018-04-19:
You can get the details of my issue here: 

https://raspberrypi.stackexchange.com/questions/82911/cant-login-to-pi0-console-from-pi3-b-via-uart-serial-usb-tty-adapter?noredirect=1#comment131471_82911

I do get a blank screen via `screen /dev/ttyUSB0 115200` but no matter how many times I press enter, a blank screen is all I see. Not sure if it is a driver related issue or what but I can't seem to figure it out.

SterlingButters, 2018-05-16:
Turns out it was some electrical issue with my Pi0

closed 2018-05-16 by SterlingButters

Rivest8, 2018-06-13:
What was the solution? I am having the same problem with the blank screen.
I found if you press CTRL + C then the "pi@raspberrypi:~$" text comes up and I am able to enter commands. So it is communicating with the Pi0 but, once I Type the test command "echo 'a' | /home/pi/sendkeys /dev/hidg0 keyboard" then it goes back to that blank screen. I cant get the computer to recognize any keyboard commands from the Pi0.

SterlingButters, 2018-06-13:
Let me know if this helps:
https://raspberrypi.stackexchange.com/questions/83209/blank-terminal-in-pi0-uart-connection?noredirect=1#comment131996_83209

Rivest8, 2018-09-11:
Thank you, Turns out the pi0 was actually working fine for me. The console is just blank until you hit any key which is ok... The issue with the keys not being recognized is actually an issue with the system I have it plugged into and not the PI KVM setup.