Fmstrat/diy-ipmi #13: Cannot connect to webserver

moose999, 2018-07-06:
Hi All,

The install seems to have gone well, but I cannot access the web site, seemingly due to a problem with lighttpd. 

A screen grab of systemctl status lighttpd.service:

[image]

Can you please help me proceed/troubleshoot?

Thanks!
Justin

Fmstrat, 2018-07-06:
What do you see from `journalctl -u lighttpd`? Have you checked the lighttpd logs?

moose999, 2018-07-06:
That got it, thanks!

the error was:

Jul 06 11:42:51 raspberrypi lighttpd[559]: Duplicate config variable in conditional 0 global: auth.debug
Jul 06 11:42:51 raspberrypi lighttpd[559]: 2018-07-06 11:42:51: (configfile.c.1154) source: /etc/lighttpd/lighttpd.conf line: 43 pos: 13 parser failed somehow near here: (EOL)

looking in /etc/lighttpd/lighttpd.conf I could see that the auth.require section was duplicated - I imagine this was my fault as I ran the installer twice due to an interruption at my end.

Many thanks!
Justin

closed 2018-07-06 by moose999