Fmstrat/diy-ipmi #14: HDMI extender as capture device

lars18th, 2018-07-24:
Hi,

This is a great project!

Regarding the current video capture method: one HDMI to TV Composed Video converter + one USB TV capture USB device. I suggest to replace both with a direct HDMI-to-Network Stream:

https://github.com/AdamLaurie/hdmi-rip
https://github.com/benjojo/de-ip-hdmi

The current solution cost around $50, that's more than the HDMI extender. Futhermore, the HDMI extender can work with all generic HDMI resolutions: 1080, 720 and SD. So it improves the flexibility if the PC to control has a common HDMI/DVI port.

DedupOperator, 2018-09-21:
Now that you mention it,
I thought of using it also for this purpose.
Already got one Lenkeng TX unit in my "some day" Todoist project storage:)

More useful links:
[Reverse Engineering Lenkeng HDMI-over-IP Extender](https://blog.danman.eu/reverse-engineering-lenkeng-hdmi-over-ip-extender/)
[newer version Lenkeng LKV373A](https://blog.danman.eu/new-version-of-lenkeng-hdmi-over-ip-extender-lkv373a/)

lars18th, 2018-09-25:
Hi @DedupOperator ,

So, I hope someone will add support for such devices in this project!

laimison, 2018-09-25:
Hi guys,

Thanks for this issue raised.
I have ordered LKV373A sender as well so cannot promise, but will have this unit in front of me and try to add support.

lars18th, 2018-09-26:
Hi @laimison , great to hear it!

spyd3rweb, 2018-11-04:
The LKV373A works with the 'diy-ipmi-video-lkv373a*" systemd service units from this latest pull request (https://github.com/Fmstrat/diy-ipmi/pull/26).  The LKV373A needs to have its firmware updated per this thread (https://www.eevblog.com/forum/reviews/lkv373a-hdmi-streamer-my-findings/msg1414171/#msg1414171).  So far the only lkv373a implementation I have working is 'diy-ipmi-video-lkv373a-video0, which is using socat to send the mpegts stream to a v4l2loopback device through ffmpeg. Lighttpd is only outputting 1-2 frames per sec; I'm not really a web guy, so I was hoping someone else would get the live video/stream playback working in html5/js.

SterlingButters, 2018-11-06:
@spyd3rweb How do you mean "working"? Join my Discord: https://discord.gg/uSTr7DZ , @DedupOperator and I have discussed some options for video streaming which may be helpful here. This was the next thing I was going to tackle in my own "fork" of this project: https://github.com/SterlingButters/ip-kvm-interface

SterlingButters, 2018-11-07:
@spyd3rweb hmmm well I was going to try to use either `ffmpeg` as well or the `motion` package and see which was better (though other alternatives were considered and I have them written down somewhere). I'm not sure where the limitation lies in the 1-2 second delay that you mention though your phrasing implies it is with `lighttpd`. As soon as I am able I will let you know if serving web elements via `express` in node.js (which I use in my project) appears faster than the `lighttpd` server.