Fmstrat/diy-ipmi #15: Comments using a KVM switch?

lars18th, 2018-07-24:
Hi,

Someone has used it (without the relay) with a regular KVM switch (with a HDMI video connection, not with the ancient VGA)?

SonicTwinkie, 2018-07-24:
Wouldn’t the KVM need to have remote switching functionality? I think you’d have to remote into two devices then, the KVM and the Pi. Or would the Pi be able to switch ports on the KVM?

lars18th, 2018-07-25:
Hi @SonicTwinkie ,

Let me to explain:

- General KVM switches have only _local_ functionality (that's local console).
- Only professional and very expensive KVM switches support IP remote access.

So the idea is to connect the Pi to the console port of a regular KVM switch. To select the input, a regular hot-key combination can be used.

I hope someone will try.

Rivest8, 2018-09-11:
I actually tried using this for that exact scenario. I have an old NTI 32 port VGA "dumb" KVM I picked up for free. Its huge but free... It has no network/IP KVM functionality at all but by plugging in the pi KVM into the monitor/USB control ports of this KVM I could theoretically control all the computers connected to it. With this one you can press CTRL + ~ then the port number and the KVM will switch control over to that port. so no need to spend a grand on an IP KVM. 

However I encountered one issue I have yet to solve. For some reason when you plug in an USB Keyboard Emulator into the keyboard port instead of a regular keyboard the KVM does not reconize any of the keystrokes. I tried this both with a an Arduino based "bad USB" rubber ducky knock off and the Pi0. It works fine plugged directly into a computer just not though a KVM. I am not sure why this is. If anyone has any ideas please let me know.

lars18th, 2018-09-12:
Hi @Rivest8 ,

Your problem with the USB Keyboard is like this...?

- When you connect to one computer, like a PC with Linux or Windows, the USB keyboard works. However, when you connect it to the USB port of the KVM then the keyboard doesn't work.

Is this what's happening to you?

Rivest8, 2018-09-12:
Correct. 
It seems to be an issue with emulated keyboard devices in general not just the pi0, with this KVM. I tried both an Pi0 and an Arduino Mega. I have a few different branded KVM's at work I can test this. I will let you know if I find one that works. 

I am also going to try tweaking the programming. Perhaps the way the KVM/PC recognizes the keyboard emulation and installs the driver has something to do with it? However I am not a strong programmer...

lars18th, 2018-09-20:
Hi @Rivest8 ,

This a common problem found in all implementations using the Linux USB Gadget HID driver: **the driver doesn't support the BOOT mode required by the KVM**.

Here the first description of the problem in the Linux mailing-list:
https://www.spinics.net/lists/linux-usb/msg124884.html

And here the commit of the patch that fixes the problem in the Linux kernel:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/usb/gadget/function/f_hid.c?id=b3c4ec71ec413c2e5bfb028bdf1737af07f1fde0

So any kernel before August 2017 suffers the same problem!

Solution: update your kernel to a recent version and recompile it. With the correct G_HID module your KVM should work.

Regards.

spyd3rweb, 2018-11-07:
I'm also looking to leverage diy-ipmi as a web/ip front end for a KVM Switch (https://www.ebay.com/itm/IOGEAR-16-Port-VGA-KVM-Miniview-Ultra-GCS1716/302811391821).

I'm able to use the Pi0 to successfully type general text through the KVM; however, I'm not able to successfully trip the OSD (ctrl, ctrl or scroll-lock, scroll-lock) or hotkey commands (num-lock + - or ctrl + F12).  Has anyone successfully used the Pi0 w/ boot mode to trigger an OSD or hotkey for a KVM?

If not I might investigate using mousejack (https://github.com/BastilleResearch/mousejack) for less nefarious purposes than it's intended. I have some vulnerable logitech unifying receivers sitting around and I know my vulnerable keyboard works with the KVM OSD/hotkey.

lars18th, 2018-11-07:
Hi @spyd3rweb ,

First of all, I **strictly** recomend to forget _all_ KVM Switches that use the \<Scroll-Lock\> Key. It's a real pain!

However, as you know the HID "boot mode" is **required** to work with KVM Switches. So, all implementations that doesn't have this patch, will never work:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/usb/gadget/function/f_hid.c?id=b3c4ec71ec413c2e5bfb028bdf1737af07f1fde0

And I don't tested this patch with a Pi0, as I prefer to use an Android device for the USB Gadget part (less expensive, as I have four old smarphones with a broken screen). If you try to support a real Android device for controlling the keyboard/mouse then I'll provide some support.

Futhermore, one question: Someone knows about HDMI KVM Switches (low cost)? I think is preferable to use only HDMI capture modules with this project, and forget the analog VGA.

SterlingButters, 2018-11-07:
So my question is (for the mouse/keyboard side of things) why are we using the Pi0 at all as opposed to Arduino? I haven't tested the arduino in "boot mode" however I've seen documentation that hints its support. Other benefits include price, no serial login, easy programming, and simplicity which gives more integrity to deployment, etc.

SterlingButters, 2018-11-07:
@lars18th You dont have to sell me on the idea of availability of options, the more the better. In fact, there are many things you can do with the Pi0 that you can't do with the arduino and I make mention of this in my repo. However, for the current implementation of only keyboard (hopefully mouse to come in my repo) makes arduino the most competitive option in my opinion. 

Additionally, I like your idea of using android, however, what would your ideas be for this in terms of remote access? My repo is targeted toward accessing the RPi3 server using DDNS and port forwarding and Im having a hard time figuring out how/where the android keyboard would fit in?

lars18th, 2018-11-10:
Hi @SterlingButters ,

> Additionally, I like your idea of using android, however, what would your ideas be for this in terms of remote access? My repo is targeted toward accessing the RPi3 server using DDNS and port forwarding and Im having a hard time figuring out how/where the android keyboard would fit in?

My idea is simple: Only a small part of programable hardware has USB slave (aka USB-gadget) support. So as practically _all_ Android devices has USB slave ports (not really true), I suggest to use it as a "remote control". Using a custom kernel, or injecting a kernel module, you can use the root shell to execute a Linux process that acts as an USB HID-Keyboard. This process can receive the keys over the network. Then if you like to use the Android device for other taks (for example to execute a VPN server) you can do it for free.

In any case, my suggestion is to decouple the HID-Keyboard emulator from the core of the IPMI. Then you need to use/define some network protocol to exchange the key strokes.

I hope the idea will be more clear now.
Regards.