Fmstrat/diy-ipmi #16: Idea: Android device as USB Keyboard/Mouse

lars18th, 2018-07-24:
Hi,

This project transforms one Android device in a USB Keyboard/Mouse:

https://github.com/pelya/android-keyboard-gadget
(used as library: https://github.com/Netdex/android-hid-script)

So, perhaps you can use one ancient rooted smartphone instead of the Pi0.

DedupOperator, 2018-09-21:
I guess it can be done but I think it's not practical (more weight and power consumption), unless you want to control the keyboard remotely by cellular network with low budget and already have the smartphone in hand.

Better use a [teensy](https://www.pjrc.com/teensy/td_keyboard.html) board.
You can also use arduino micro but I'm not sure there is a ready made library that support using keyboard in BIOS.

lars18th, 2018-09-25:
Hi @DedupOperator .

Well form my point of view it will be practical if this project be "customizable". Perhaps someone likes to use for the USB HID interface: a) Raspberry Pi Zero; b) Teensy board; c) Android device.

Why not support different options?

DedupOperator, 2018-09-25:
@lars18th I guess you're right.
The project can expend, but I think some more details about use cases will help focus and priority.

Check out this little device it works on android out of box [InputStick](http://inputstick.com/)

lars18th, 2018-09-26:
Hi @DedupOperator ,

This InputStick seems to be interesting. However, the idea of using an Android device is because tipycally it has support for the USB Gadget library. So it can impersonate an HID device. The Raspberry Pi Zero has this function too; but __not__ the regular Pi. Thus, the concept is to support different hardware "options" for the connection between the main "diy-ipmi" controller and the USB port of the target device (the one commonly used when connecting the USB keyboard/mouse).

Regards.

Fmstrat, 2018-09-26:
As stated in other tickets, happy to merge a PR for anything, though I'm not sure why you would want an expensive Android device to replace a $5 part (minus mouse control).

lars18th, 2018-09-27:
> though I'm not sure why you would want an expensive Android device to replace a $5 part (minus mouse control).

Because the cost of an 4-5 years old Android device with a broken screen is nearly... zero? :smile: 

Now, seriously: My idea is to run the **Manager** (the central point) in a docker container inside a _shared_ x86 server; instead of run it in a Raspberry Pi board. So the **Relay board** isn't required. The **Video capture** can be done using some HDMI-2-Network device. And the **HID USB module** can be best done over the Network with a (reused) Android device.

My suggestion targets the "modularity" of the project.
I hope you agree with that!
Regards.