Fmstrat/diy-ipmi #18: Will it work only as remote keyboard? So no errors will be thrown when Easycap device is not connected.

laimison, 2018-07-29:
It will be great if someone can test it. Many thanks for the response and great & rarely found solution!

Fmstrat, 2018-07-31:
I'm not sure what you're asking here. It certainly works as a remote keyboard. Is that what you're asking?

laimison, 2018-07-31:
Many thanks for response. Let's say I don't need video streaming. Is the tool capable to ignore the fact that I do not have components required for video streaming? So basically it has three functionalities 1) remote keyboard 2) reboot button relay 3) video streaming - can I use just one of them and script will work fine?

Fmstrat, 2018-07-31:
Yes, it could. However, if you look through the scripts, you could do that with just a Pi 0 W and not use a full size Pi at all.

laimison, 2018-08-06:
Thanks. That helped a lot. Now I believe that these three functionalities are isolated in a code so it's even easier to debug if any issue.

laimison, 2018-08-11:
I very enjoy this tool. Today I have set up keyboard successfully as per your notes that only Pi Zero W should be enough. I don't know where the proper place is to share my experience, but recommended http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-03-03/ didn't work. It was no way to get /dev/hidg0 available. Possible bug was discovered here with 4.4.50+ kernel - https://github.com/aidantwoods/RPi0w-keyboard/issues/1#issuecomment-309244795 . I just ended up by downgrading to 
http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2016-11-29/2016-11-25-raspbian-jessie-lite.zip successfully! :)

closed 2018-09-01 by laimison