Fmstrat/diy-ipmi #20: HDMI to S-Video glitches with mencoder

laimison, 2018-09-01:
Hi,

I have tested

- two different converters:

https://www.ebay.co.uk/itm/Mini-Portable-USB-2-0-Port-HD-1-Way-HDMI-1080P-Video-Capture-Card-For-PC-Laptop/142907175902?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

and

https://www.ebay.co.uk/itm/TV-DVD-VHS-RCA-S-Video-Video-Audio-AV-EasyCAP-Grabber-Capture-USB-2-0-Adapter/183096373306?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649
+
https://www.ebay.co.uk/itm/HDMI-to-3RCA-AV-CVBS-Composite-S-Video-R-L-Audio-Converter-Adapter-Upscaler/262370360688?epid=1140068747&hash=item3d167dfd70:g:gZYAAOSwCpRbEhZ7

- two different HDMI cables
- two different computers to output HDMI (PC with GTX1070 and Macbook Pro 2014)
- Raspberry Pi3 B+ with latest Raspbian Stretch 9 (Linux raspberrypi 4.14.50-v7+ #1122 SMP Tue Jun 19 12:26:26 BST 2018 armv7l GNU/Linux)

I've got the same result by using any of these converters and cables.

I'm using mencoder command to record test.avi and the output is almost not visible because of many green lines and glitches:

[image]

This is a mencoder command which prints some drops (`14 frames successfully processed, -10 frames dropped`):
```
pi@raspberrypi:~ $ mencoder tv:// -tv driver=v4l2:norm=NTSC:device=/dev/video0:input=1:fps=5 -nosound -ovc copy -o test.avi
MEncoder 1.3.0 (Debian), built with gcc-6.2.1 (C) 2000-2016 MPlayer Team
success: format: 9  data: 0x0 - 0x0
TV file format detected.
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
Selected device: usbtv
 Capabilities:  video capture  read/write  streaming
 supported norms: 0 = NTSC; 1 = NTSC-M; 2 = NTSC-M-JP; 3 = NTSC-M-KR; 4 = NTSC-443; 5 = PAL; 6 = PAL-BG; 7 = PAL-H; 8 = PAL-I; 9 = PAL-DK; 10 = PAL-M; 11 = PAL-60;
 inputs: 0 = Composite; 1 = S-Video;
 Current input: 0
 Current format: YUYV
Selected input hasn't got a tuner!
v4l2: ioctl set mute failed: Invalid argument
[V] filefmt:9  fourcc:0x32595559  size:720x480  fps:5.000  ftime:=0.2000
videocodec: framecopy (720x480 24bpp fourcc=32595559)
Forcing audio preload to 0, max pts correction to 0.
v4l2: select timeout

Skipping frame!
Pos:   0.0s      1f ( 0%)  0.95fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.0s      2f ( 0%)  1.27fps Trem:   0min   0mb  A-V:0.000 [0:0]
v4l2: select timeout

Skipping frame!
Pos:   0.0s      3f ( 0%)  1.43fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.0s      4f ( 0%)  1.53fps Trem:   0min   0mb  A-V:0.000 [0:0]
v4l2: select timeout

Skipping frame!
Pos:   0.0s      5f ( 0%)  1.59fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.0s      6f ( 0%)  1.85fps Trem:   0min   0mb  A-V:0.000 [0:0]
Writing header...
ODML: Aspect information not (yet?) available or unspecified, not writing vprp header.
Writing header...
ODML: Aspect information not (yet?) available or unspecified, not writing vprp header.
Pos:   0.0s      7f ( 0%)  2.13fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.2s      8f ( 0%)  2.40fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.2s      9f ( 0%)  2.67fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.2s     10f ( 0%)  2.93fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.2s     12f ( 0%)  3.43fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.4s     13f ( 0%)  3.68fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.4s     14f ( 0%)  3.92fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.4s     15f ( 0%)  4.15fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.4s     17f ( 0%)  4.61fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.6s     18f ( 0%)  4.82fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.6s     19f ( 0%)  5.03fps Trem:   0min   0mb  A-V:0.000 [0:0]

Invalid frame duration value (0.520/0.000 => -0.520). Defaulting to 0.040 sec.

Skipping frame!
Pos:   0.6s     20f ( 0%)  4.65fps Trem:   0min   0mb  A-V:0.000 [0:0]
v4l2: select timeout

Skipping frame!
Pos:   0.6s     21f ( 0%)  4.35fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.6s     22f ( 0%)  4.11fps Trem:   0min   0mb  A-V:0.000 [0:0]
v4l2: select timeout

Skipping frame!
Pos:   0.6s     23f ( 0%)  3.92fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.6s     24f ( 0%)  3.75fps Trem:   0min   0mb  A-V:0.000 [0:0]
v4l2: select timeout

Skipping frame!
^Cs:   0.6s     25f ( 0%)  3.61fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.6s     26f ( 0%)  3.50fps Trem:   0min   0mb  A-V:0.000 [0:0]
Writing index...
Writing header...
ODML: Aspect information not (yet?) available or unspecified, not writing vprp header.

Video stream: 27648.000 kbit/s  (3456000 B/s)  size: 2073600 bytes  0.600 secs  26 frames
v4l2: select timeout
v4l2: ioctl set mute failed: Invalid argument
v4l2: 14 frames successfully processed, -10 frames dropped.
pi@raspberrypi:~ $
```

Can you see guys an issue or do you need more details? I really appreciate any comment on that - thanks!

laimison, 2018-09-05:
I have found that removing `modules-load=dwc2` from `/boot/cmdline.txt` and `dtoverlay=dwc2` from `/boot/config.txt` seems sorted the root cause in my case. I came to idea to remove this because of `[253805.452822] dwc2 3f980000.usb: dwc2_hc_halt() Channel can't be halted` error in `dmesg`. 

I still have two issues that
1) video quality is not good enough to see very small fonts (it's like 30+ years TV display)
I still see some `buffer contains corrupted data` when using ffmpeg or `avconv` tools

2) and lower corner of screen is not visible through web server (so when I'm in the Linux console, I don't see few last lines what I'm typing in)

Now sure whether these next issues are related with original one at the moment

laimison, 2018-09-13:
I wanted to share a resolution.

A second issue solved by changing `avconv -f video4linux2 ...` to `avconv -f v4l2 ...` in `/var/www/html/image.php` so it fits into the screen now.

Video/image quality still could be better, but decided to not invest time on this issue.

An example with "HD Capture":
[image]

DedupOperator, 2018-09-21:
Unfortunately, analog video has a very low quality.
From my past experience there are EasyCap devices that use bad qualiy ADC chip, device looks the same but different chips inside.
The Amazon link provided should be the better quality one.

You can research more but I think the best way is going fully digital, you can check issue #6

DedupOperator, 2018-09-25:
@laimison refer issue #14 for an interesting "hackish" way to capture HDMI signals with minimal budget if you would like to upgrade your quality.

About the cloned devices, I meant the Easycap device.
But anyway you can't expect much from those cheapo product.
The downscaling from HDMI resolutions, to analog TV resolutions like S-Video will give you approximately the equivalent of 640x480-800x600 resoluion.
Text won't be sharp in this technology unless higher quality analog-to-digital and digital-to-analog converters are used.

Best option is to use an all-digital path as I mentioned before.