Fmstrat/diy-ipmi #21: Is it possible to go to the Bios with this fake keyboard?

laimison, 2018-09-13:
Hi,

I have noticed that fake keyboard driver is not working when the computer is rebooting and I press "F2".

```
pi@raspberrypi:~$ echo "left-ctrl left-alt delete" | /home/pi/sendkeys /dev/hidg0 keyboard
recv report: 00
recv report: 01
pi@raspberrypi:~$ echo 'f2' | /home/pi/sendkeys /dev/hidg0 keyboard
pi@keyboardrypi:~$ echo 'f2' | /home/pi/sendkeys /dev/hidg0 keyboard 
^C
pi@raspberrypi:~$ echo 'f2' | /home/pi/sendkeys /dev/hidg0 keyboard
^C
pi@raspberrypi:~$ echo 'f2' | /home/pi/sendkeys /dev/hidg0 keyboard
^C
pi@raspberrypi:~$ echo 'f2' | /home/pi/sendkeys /dev/hidg0 keyboard
^C
pi@raspberrypi:~$ echo 'f2' | /home/pi/sendkeys /dev/hidg0 keyboard
^C
pi@raspberrypi:~$ echo 'f2' | /home/pi/sendkeys /dev/hidg0 keyboard
recv report: 00
^C
pi@raspberrypi:~$ 
pi@raspberrypi:~$ echo 'f2' | /home/pi/sendkeys /dev/hidg0 keyboard
recv report: 01
^C
pi@raspberrypi:~$ echo 'f2' | /home/pi/sendkeys /dev/hidg0 keyboard
^C
pi@raspberrypi:~$ echo 'f2' | /home/pi/sendkeys /dev/hidg0 keyboard
recv report: 01
```

If I press "F2" with normal keyboard, then after a moment this fake keyboard becomes available.

Do you guys can confirm that BIOS worked for you? Maybe this related with Raspberry Pi0 powering? (so Pi0 power is interrupted by the computer during reset moment)

It will be really helpful at least to know whether this issue is specific to my hardware/case so you can get to BIOS remotely.

SterlingButters, 2018-09-13:
BIOS worked for me

laimison, 2018-09-13:
Thank you, that is useful information. Can you let me know your mainboard model and Pi0 build?
My one is Strix z270e Gaming Intel z270 lga1151 ATX and Pi0 build details:

```
pi@raspberrypi:~$ uname -a
Linux raspberrypi 4.4.34+ #930 Wed Nov 23 15:12:30 GMT 2016 armv6l GNU/Linux
pi@raspberrypi:~$ cat /etc/*release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~$
```

So I can do the rest find out the difference between our configurations.
I'm guessing that USB cable between Pi0 and computer doesn't matter. I have tried few of them just in case.

SterlingButters, 2018-09-13:
Unfortunately I don't have access to my Pi0 at the moment and wont for several weeks but this is my motherboard: ASRock X299 Taichi LGA 2066 Intel X299 SATA 6Gb/s USB 3.1 ATX Intel Motherboard

laimison, 2018-09-13:
Thanks a lot! I have ordered additional SD card for Pi Zero so will try this image and some different computers.

Fmstrat, 2018-09-14:
How are you powering the Pi0? Is it a Pi0 or Pi0w? If you power everything except the PC up, wait for Pi booting, then turn on PC from cold, does that work?

SterlingButters, 2018-09-14:
Glad you got it working though I wanted to check my knowledge against @Fmstrat that the “port concept” you don’t understand concerning the PC-Pi0 connection is for the HID emulation (i.e. you are correct when you said “data transfer”). Though, I can’t say I completely understand the way Pi’s handle power because my Pi3 can somehow run off of an HDMI!!!

laimison, 2018-09-23:
Yes, I have noticed this as well related with HDMI to USB powering.

In terms of PC-Pi0 connection, there is one more finding. I just cut 'red' wire and it worked well after some restarts when testing this. It's better solution because Pi Zero doesn't depend on BIOS parameter.

seffyroff, 2018-11-09:
I have the same problem here - all of the machines I've connected up aren't taking input during bios POST, or even until the OS has fully booted, so accessing OS rescue modes and so on is off limits also.  Not sure what else I can do here, is there some BIOS setting that'll help?  The boxes are all older, but not that old. A regular USB keyboard works in the BIOS.  Is there something in the Pi0 USBHID configuration that could be tweaked?  I've got all the Pis and the vidcap powered separately via a powered hub.

laimison, 2018-11-09:
Check if you have ErP or EuP enabled in APM section. Usually, it's not by default.

I follow this procedure:

1) Powering off the machine
2) Trying to connect to Pi0 via serial (if you are lucky, you can probably jump to step 6, if not see the next step)
3) Disconnecting HID keyboard with relay - custom step
4) Restarting Pi0 by cutting power with relay - custom step
5) Connecting HID keyboard with relay - custom step
6) Powering on the machine
7) Connecting to Pi0 and doing `echo 'delete' | /home/pi/sendkeys /dev/hidg0 keyboard`

As you see in steps 3 and 4, I'm able to disconnect keyboard and turn off Pi0 remotely because I routed these wires through separate relay's channels.

I believe that main issue is that your Pi0 is powered from your machine through USB. So for some people BIOS is waiting less time than Pi0 is able to boot up.

Theoretically faster storage on Pi0 could solve this issue, if it would be very fast. So in real world I don't think this is the solution. I had to go through these 7 steps to make sure 100% that I will be able to reach BIOS remotely.

laimison, 2018-11-10:
If I understand correctly usually Pi0 is powered from the source which was connected first.

So there should be a possibility that your machine powers Pi0 through USB as any normal keyboard in the world, because it may reached Pi0 first than normal power source. It may not.

You can see how this works by attaching to serial mode `screen /dev/ttyUSB0 115200` from Pi3 to Pi0, leaving this screen as it is and trying to cut these two power sources in different orders.

EuP may help you so it's worth to test that machine. If there is no luck, I would still play with cables attached to Pi0 (USB and power cable) and powering your machine on/off.

Your Pi0 is maybe crashing or rebooting at some event

I'm trying to brainstorm and I do not know about fast boot, but maybe it can delay bios loading which is quick thing to try as well.

If there is no luck, I'm out of ideas.

I would try different laptops or PCs that I have to confirm that Pi0 keyboard cannot talk with any BIOS. Then you can focus on Pi0 configuration/build. Not all Jessie builds are good.

laimison, 2018-12-02:
Just to not expect an answer from me for your plan, unfortunately I didn't get deeper into driver issue. In my case I have solved this issue for hardware which is not older than 3 years.