Fmstrat/diy-ipmi #23: screen recording issue

pan87232494, 2018-10-29:
I folllowed your guide, and copied hardware setting
https://www.amazon.com/dp/B012MDMWLM
https://www.amazon.com/dp/B0126O0RDC

I want to use this to connect to my ubuntu desktop machine ( 1080ti + intel 8700 core video chip)
but I always get black screen, when I do test via following command:

mencoder tv:// -tv driver=v4l2:norm=NTSC:device=/dev/video0:input=1:fps=5 -nosound -ovc copy -o test.avi
avconv -f video4linux2 -i /dev/video0 -vframes 1 -s 720x480 test.jpg

and also, I get alot frame skiping message, please help. thanks a lot.

--------------------------------env info-------------------------
pi@raspberrypi:~ $ mencoder -v
MEncoder 1.3.0 (Debian), built with gcc-6.2.1 (C) 2000-2016 MPlayer Team

pi@raspberrypi:~ $ mencoder tv:// -tv driver=v4l2:norm=NTSC:device=/dev/video0:input=1:fps=5 -nosound -ovc copy -o test.avi
MEncoder 1.3.0 (Debian), built with gcc-6.2.1 (C) 2000-2016 MPlayer Team
success: format: 9  data: 0x0 - 0x0
TV file format detected.
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
Selected device: usbtv
 Capabilities:  video capture  read/write  streaming
 supported norms: 0 = NTSC; 1 = NTSC-M; 2 = NTSC-M-JP; 3 = NTSC-M-KR; 4 = NTSC-443; 5 = PAL; 6 = PAL-BG; 7 = PAL-H; 8 = PAL-I; 9 = PAL-DK; 10 = PAL-M; 11 = PAL-60;
 inputs: 0 = Composite; 1 = S-Video;
 Current input: 1
 Current format: YUYV
Selected input hasn't got a tuner!
v4l2: ioctl set mute failed: Invalid argument
[V] filefmt:9  fourcc:0x32595559  size:720x480  fps:5.000  ftime:=0.2000
videocodec: framecopy (720x480 24bpp fourcc=32595559)
Forcing audio preload to 0, max pts correction to 0.
Writing header...
ODML: Aspect information not (yet?) available or unspecified, not writing vprp header.
Writing header...
ODML: Aspect information not (yet?) available or unspecified, not writing vprp header.
Pos:   0.0s      1f ( 0%)  0.00fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.2s      2f ( 0%)  0.00fps Trem:   0min   0mb  A-V:0.000 [0:0]

Skipping frame!
Pos:   0.2s      3f ( 0%)  0.00fps Trem:   0min   0mb  A-V:0.000 [0:0]
.....

DedupOperator, 2018-10-30:
Hi @pan87232494 I recommend reading the recent updates of the issues, there is some usuful information over there.

Also, can you please try:
`ffplay -f video4linux2 -list_formats all /dev/video0`

you might need the ffmpeg package.

pan87232494, 2018-10-30:
many thanks for your replay 
here is output
pi@raspberrypi:~ $ ffplay -f video4linux2 -list_formats all /dev/video0
ffplay version 3.2.10-1~deb9u1+rpt2 Copyright (c) 2003-2018 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
  configuration: --prefix=/usr --extra-version='1~deb9u1+rpt2' --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libebur128 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx-rpi --enable-mmal --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libiec61883 --arch=armhf --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  WARNING: library configuration mismatch
  avutil      configuration: --prefix=/usr --extra-version='1~deb9u1+rpt1' --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libebur128 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx-rpi --enable-mmal --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  avcodec     configuration: --prefix=/usr --extra-version='1~deb9u1+rpt1' --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libebur128 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx-rpi --enable-mmal --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  avformat    configuration: --prefix=/usr --extra-version='1~deb9u1+rpt1' --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libebur128 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx-rpi --enable-mmal --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  avfilter    configuration: --prefix=/usr --extra-version='1~deb9u1+rpt1' --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libebur128 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 - [... cut ...]
  libavcodec     57. 64.101 / 57. 64.101
  libavformat    57. 56.101 / 57. 56.101
  libavdevice    57.  1.100 / 57.  1.100
  libavfilter     6. 65.100 /  6. 65.100
  libavresample   3.  1.  0 /  3.  1.  0
  libswscale      4.  2.100 /  4.  2.100
  libswresample   2.  3.100 /  2.  3.100
  libpostproc    54.  1.100 / 54.  1.100

[video4linux2,v4l2 @ 0x68300490] Raw       :     yuyv422 :           YUYV 4:2:2 :
/dev/video0: Immediate exit requested
    nan    :  0.000 fd=   0 aq=    0KB vq=    0KB sq=    0B f=0/0

pan87232494, 2018-10-30:
I tried one more time using your install.sh to setup on clean env, raspbian image and boot/config, keep the same.
pi 2 --- USB TTL Serial cable----- pi0 ---------micro usb ------ pi0 (as client)
   |                                                                                                  |
   ------------------easycap--------------s video convert-----------

then I see green lines get displayed, but still not working, and I get black image from test output

[image]

pan87232494, 2018-10-30:
plus, 
1. this i syslog message is seen, then putty get no respone from pi2.
Message from syslogd@raspberrypi at Oct 30 07:05:15 ...
 kernel:[ 1108.269092] Process avconv (pid: 10373, stack limit = 0x8fa30210

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
12239 www-data  20   0  126980  19724  15844 R  17.7  2.2   0:00.54 avconv
 1721 pi        20   0    8220   3280   2720 S   2.0  0.4   0:23.91 top
11310 pi        20   0    8100   3244   2764 R   1.3  0.4   0:01.90 top

pan87232494, 2018-10-30:
pi@raspberrypi:/var/www/html $ dmesg | grep usb
....
[    2.487546] usb 1-1.3: Product: usbtv007
[    2.487554] usb 1-1.3: Manufacturer: fushicai
[    2.487562] usb 1-1.3: SerialNumber: 300000000002
[    2.691518] usb 1-1.1.1: new high-speed USB device number 7 using dwc_otg
[    2.821895] usb 1-1.1.1: New USB device found, idVendor=0424, idProduct=7800
[    2.821906] usb 1-1.1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    7.559242] usbcore: registered new interface driver brcmfmac
[    7.626431] usbcore: registered new interface driver usbserial
[    7.626527] usbcore: registered new interface driver usbserial_generic
[    7.626582] usbserial: USB Serial support registered for generic
[    7.632313] usbcore: registered new interface driver pl2303
[    7.632395] usbserial: USB Serial support registered for pl2303
[    7.642579] usb 1-1.2: pl2303 converter now attached to ttyUSB0
[    7.801011] usbtv 1-1.3:1.0: Fushicai USBTV007 Audio-Video Grabber
[    7.801208] usbcore: registered new interface driver usbtv
[    7.801011] usbtv 1-1.3:1.0: Fushicai USBTV007 Audio-Video Grabber
[    7.801208] usbcore: registered new interface driver usbtv
[   60.415412] usbtv 1-1.3:1.0: Could not resubmit ISO URB
[   64.375293] usbtv 1-1.3:1.0: Could not resubmit ISO URB
[   85.371290] usbtv 1-1.3:1.0: Could not resubmit ISO URB
[   94.391290] usbtv 1-1.3:1.0: Could not resubmit ISO URB
[  103.415102] usbtv 1-1.3:1.0: Could not resubmit ISO URB
[  110.317281] usbtv 1-1.3:1.0: Could not resubmit ISO URB
[  154.276294] usbtv 1-1.3:1.0: Could not resubmit ISO URB
[  161.777189] usbtv 1-1.3:1.0: Could not resubmit ISO URB
Could not resubmit ISO URB

pan87232494, 2018-10-30:
sudo apt-get update
sudo apt-get dist-upgrade
sudo apt-get install dkms
reboot
and mplayer tv:// -tv device=/dev/video0 -hardframedrop will see video come out.
after this I get video displayed.  but seems keyboard is not pass through. debuging

pan87232494, 2018-10-30:
BTW,  when I watched browser console, when I press key, there is no post request to httpserver.
is these some steps I missed? 

<?php
	$tty = $_POST['tty'];
	$key = $_POST['k'];
	$cmd='echo "echo \''.$key.'\' | /home/pi/sendkeys /dev/hidg0 keyboard" >> '.$tty;
	system($cmd);
?>

Fmstrat, 2018-10-30:
What is in your `/etc/ipmi.conf`? And have you tried sending keypresses from the command line (see README)?

Fmstrat, 2018-11-01:
OK, that's good then. Something is probably showing up in the PHP logs. Can you tail them while attempting the commands? And just to be sure, you are "typing" on the webpage, not "clicking" on the letters, right?