Fmstrat/diy-ipmi #28: [Suggestion] Ease up deployment with some Containerization

seffyroff, 2018-11-06:
Ease the deployment with a Docker image?  I can see it being non-trivial due to the various interconnected services and whatnot, so it bears some thought on what the approach might be.  I'm interested in taking a stab at this, with some opinions shared on implementation.

Every project on GH has a Dockerfile these days :)

I also wondered about Snappy packaging.  This is a much more involved process but if done right can make deployment management trivially simple.  I've dabbled a little in this, and it's wonderful in my experience.

Fmstrat, 2018-11-06:
Yea, almost all of my projects have been in docker (see my repos) but this really didn't make sense to me as it was supposed to be a hardware IPMI via the Pi. It would be fairly trivial to create using the script.

OrbitalSystems, 2019-08-08:
any movement on this ?