Fmstrat/diy-ipmi #30: [Suggestion] Monitor current power state

seffyroff, 2018-11-06:
Related to #29 - monitor the current power state of the managed systems,  This could be done by adding a loop to the power LED of the systems and hooking up to the Pi0 GPIO, for example.

spyd3rweb, 2018-11-10:
I just posted up pics (https://github.com/Fmstrat/diy-ipmi/issues/8#issuecomment-437565628) of the custom y-splitter cables I ended up making for the power switch relay, which included an additional set of wires for the power led.  The voltage measures 2.742v at power on for my Dell T1700's.

You're right according to this thread (https://www.raspberrypi.org/forums/viewtopic.php?t=27426), the Pi GPIO will read 2.5v as high, essentially getting you the needed 1 bit ADC.

If more voltage or channel flexibility is needed there appears to be several ADC HATs for the pi:
- 8 channel, 10 bit: (https://hackaday.com/2017/09/22/8-channel-adc-for-the-raspberry-pi/)
- 12 channel, 10 bit: (https://www.tindie.com/products/cburgess129/12-channel-1012-bit-adc-hat-for-raspberry-pi/)
- 16 channel, 16 bit, -2.5v to +2.5v range: (https://www.amazon.com/Alchemy-Power-Inc-Pi-16ADC-Converter/dp/B01N7J31C1)

seffyroff, 2018-11-11:
I'm looking at adding in some frontend state indicators, and also some support for my HDMI splitter box (I didn't want to use multiple capture cards, and this can be IR controlled by a NodeMCU/Lirc USB from Pi).