Fmstrat/diy-ipmi #32: Even cheaper KVM with only Pi Zero W

JsBergbau, 2019-02-25:
Hallo Fmstrat,

thanks for providing this full featured KVM solution.
I didn't find any contact E-Mail so I use this issue section.

Have you ever thought about making this KVM even a lot cheaper? With a raspberry PI Zero W and the Raspi Camera you only need those 2 devices (and a monitor of course). The kamera just records the screen, sends it and you can also interact. No need for a RPi 3, Videograbber and stuff like that.
Of course your code is opensource, everyone can adopt your code. However you know your code, you probably would to it in a tenth of the time someone would need who never had to do with your code before.

Can you create an additional project with KVM based solo on a raspbery PI Zero? Together with TP-Link HS110 you can even switch the PC on and off https://github.com/softScheck/tplink-smartplug

Thanks in advance.

crankmosh, 2019-05-10:
that defeats the whole purpose, now you need a monitor to point the camera to

JsBergbau, 2019-05-23:
I would not say it defeats the whole purpose. It is a nice extra for example to input password for pre-boot authentification. After that you can access the PC via RDP.

jbrepogmailcom, 2019-10-15:
Hello I completely agree and vote for the cheap solution. I am a consultant working for several clients. They all give me their own laptop with all high security, so I can not control it with TeamViewer or similar. Yet I need to control 3 or 4 laptops. Hardware KVM even with camera would be perfect for that.

auctionsjeff, 2019-11-06:
Building off of [this post](https://sudonull.com/posts/5780-IP-KVM-DIY-30), who cites [this blog](https://blog.danman.eu/new-version-of-lenkeng-hdmi-over-ip-extender-lkv373a/), why not "watch" the RTSP stream produced by [this HDMI to TCP transmitter](https://www.amazon.com/dp/B07RZ54PRQ) ($39) and use the TTL keyboard/mouse from this repo?