Fmstrat/diy-ipmi #33: USB Gadget Problems

seffyroff, 2019-03-01:
I'm unable to use the keyboard functionality of diy-ipmi.  After poking around a bunch I'm no clearer on what the issue is.  I can see in the process list that if I've tried sending a key the sendkey process is sat there indefinitely.  I also see that there are 3 agetty processes (might be fine?) and one of them is sucking up almost 100% CPU:

[image]

seffyroff, 2019-03-01:
I also note in dmesg the following events:
`[   24.410034] dwc2 20980000.usb: dwc2_hsotg_enqueue_setup: failed queue (-11)
[   24.413187] dwc2 20980000.usb: bound driver configfs-gadget
`

seffyroff, 2019-03-01:
Hmm, it looks like @Fmstrat had the same issues [a long time ago](https://www.raspberrypi.org/forums/viewtopic.php?t=195500).  Did you figure this out?

Fmstrat, 2019-03-01:
I believe that issue went away for me when I went to a Pi0W, and when I tried another Pi0 it never showed back up. Sorry I can't provide any more specific information.

seffyroff, 2019-03-01:
Well, I'm using Pi0Ws here.  In a moment of frustration I apt-get updated the Pi0s and now there's a crash logged in dmesg on boot and /dev/hidg0 has disappeared.  I tried to recompile sendkeys and hub-ctrl in case some lib updates had broken them but that doesn't seem to be it.

Here's the crash:

```
[   17.545344] Unable to handle kernel NULL pointer dereference at virtual address 00000002
[   17.556835] pgd = d8528000
[   17.561249] [00000002] *pgd=1a6d2831, *pte=00000000, *ppte=00000000
[   17.569417] Internal error: Oops: 17 [#1] ARM
[   17.575553] Modules linked in: usb_f_hid hci_uart btbcm bluetooth brcmfmac brcmutil snd_bcm2835 cfg80211 snd_pcm rfkill snd_timer snd bcm2835_gpiomem uio_pdrv_genirq uio fixed libcomposite dwc2 udc_core ipv6
[   17.600006] CPU: 0 PID: 599 Comm: ls Not tainted 4.9.35+ #1014
[   17.607859] Hardware name: BCM2835
[   17.613223] task: d8af5f60 task.stack: d88cc000
[   17.619852] PC is at alloc_ep_req+0x30/0xa8 [libcomposite]
[   17.627442] LR is at 0x0
[   17.631984] pc : [<bf0b6ff0>]    lr : [<00000000>]    psr: a0000013
sp : d88cdd98  ip : 00000000  fp : d88cddb4
[   17.647688] r10: da56bac8  r9 : d89382c4  r8 : da56ba70
[   17.654980] r7 : d89382a8  r6 : d8a79100  r5 : da616290  r4 : 00000008
[   17.663649] r3 : d8a79134  r2 : 00000000  r1 : 00000000  r0 : d8a79100
[   17.672262] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   17.681596] Control: 00c5387d  Table: 18528008  DAC: 00000055
[   17.689513] Process ls (pid: 599, stack limit = 0xd88cc188)
[   17.697255] Stack: (0xd88cdd98 to 0xd88ce000)
[   17.703662] dd80:                                                       00000001 da56ba70
[   17.715979] dda0: 00000000 bf3c9c80 d88cddec d88cddb8 bf3c87f0 bf0b6fcc d8917cc0 d8938324
[   17.728302] ddc0: d88cde04 d88cddd0 bf0b237c da56ba70 d89382a8 da6f85e0 d893831c da56ba70
[   17.740633] dde0: d88cde04 d88cddf0 bf0b20c0 bf3c875c d89382a8 da6f868c d88cde3c d88cde08
[   17.753090] de00: bf0b638c bf0b2050 da59dd10 da6f8654 c013ef88 da686600 bf07bea4 da6f85e0
[   17.765763] de20: da6f85e0 bf07be98 00000000 d85cf000 d88cde5c d88cde40 bf079838 bf0b614c
[   17.778587] de40: 00000000 da686600 bf07bea4 da7995e0 d88cde84 d88cde60 bf079d68 bf079808
[   17.791574] de60: 00000032 da7995e0 d8a79e80 da6f8400 da6f8590 0000000d d88cdeac d88cde88
[   17.804835] de80: bf0b6d18 bf079c74 0000000d d8a79e80 d8a79e98 d88cdf80 d8a57e60 b6f19000
[   17.818209] dea0: d88cdedc d88cdeb0 c01bb3b8 bf0b6c74 d88cdf80 d8a57e60 d88cdf80 b6f19000
[   17.831732] dec0: d88cdf80 0000000d d88cc000 00000000 d88cdf4c d88cdee0 c0144244 c01bb2f0
[   17.845412] dee0: b6f19000 d88cdfb0 01271df8 00000000 d88cdfac d88cdf00 c0009204 c05d4f30
[   17.859161] df00: 5c79b46f 3a694963 0000287f da783554 00000003 00001000 00000000 c0145040
[   17.872968] df20: d88cc000 c0147778 d88cdf4c 0000000d d8a57e60 b6f19000 d88cdf80 c000ffc4
[   17.886787] df40: d88cdf7c d88cdf50 c014507c c0144218 00000001 c01624c4 d8a57e60 d8a57e60
[   17.900604] df60: 0000000d b6f19000 c000ffc4 d88cc000 d88cdfa4 d88cdf80 c014618c c0144fd8
[   17.914423] df80: 00000000 00000000 0000000d b6f19000 b6e90b40 00000004 00000000 d88cdfa8
[   17.928183] dfa0: c000fe40 c0146144 0000000d b6f19000 00000001 b6f19000 0000000d 00000000
[   17.941943] dfc0: 0000000d b6f19000 b6e90b40 00000004 0000000d b6f19000 0000000d 00000000
[   17.955757] dfe0: 00000000 bed2e83c b6dbe008 b6e1589c 60000010 00000001 00000000 00000000
[   17.969688] [<bf0b6ff0>] (alloc_ep_req [libcomposite]) from [<bf3c87f0>] (hidg_bind+0xa0/0x258 [usb_f_hid])
[   17.985266] [<bf3c87f0>] (hidg_bind [usb_f_hid]) from [<bf0b20c0>] (usb_add_function+0x7c/0x164 [libcomposite])
[   18.001224] [<bf0b20c0>] (usb_add_function [libcomposite]) from [<bf0b638c>] (configfs_composite_bind+0x24c/0x360 [libcomposite])
[   18.018869] [<bf0b638c>] (configfs_composite_bind [libcomposite]) from [<bf079838>] (udc_bind_to_driver+0x3c/0xe0 [udc_core])
[   18.036100] [<bf079838>] (udc_bind_to_driver [udc_core]) from [<bf079d68>] (usb_gadget_probe_driver+0x100/0x14c [udc_core])
[   18.053165] [<bf079d68>] (usb_gadget_probe_driver [udc_core]) from [<bf0b6d18>] (gadget_dev_desc_UDC_store+0xb0/0xcc [libcomposite])
[   18.070992] [<bf0b6d18>] (gadget_dev_desc_UDC_store [libcomposite]) from [<c01bb3b8>] (configfs_write_file+0xd4/0x17c)
[   18.087477] [<c01bb3b8>] (configfs_write_file) from [<c0144244>] (__vfs_write+0x38/0x128)
[   18.101323] [<c0144244>] (__vfs_write) from [<c014507c>] (vfs_write+0xb0/0x1bc)
[   18.111602] [<c014507c>] (vfs_write) from [<c014618c>] (SyS_write+0x54/0xb0)
[   18.121575] [<c014618c>] (SyS_write) from [<c000fe40>] (ret_fast_syscall+0x0/0x1c)
[   18.134621] Code: ebff044b e2506000 0a000013 e5952024 (e1d230d2) 
[   18.143686] ---[ end trace 557d79704a7af152 ]---

```

seffyroff, 2019-03-02:
I used rpi-update to revert back to an older kernel and now I'm back at the original problem as stated in the original post.  Take note anyone wondering if you can update to a newer kernel/Raspbian version.  Don't!