Fmstrat/diy-ipmi #40: KVM with only Pi Zero W

jtoland, 2019-09-03:
Hi,

I don't know if you are intending to update this but I am currently looking at getting this running on a single pi zero w. You can now get a CSI to HDMI adaptor so that everything can just run off 1 pi.

I was wondering if you could help me get this working if you have some spare time.

Kind Regards

SterlingButters, 2019-09-04:
@jtoland myself and some collaborators are currently working on this. Hope to have some results soon. Will also be adding mouse and mass storage support. However you may find that a pi zero may not have sufficient computational capacity to process audio/video. Since the Pi4 supports OTG, this is a better option.

jtoland, 2019-09-05:
Hello,

Thank you for confirming that you are doing this with the pi 4.

i have tested video on the pi zero and this seems to work fine with keyboard but having mouse working as well i can can see where the bottleneck could be.

Do you have any kind of time frame on the pi 4 version as i would love to test this even if it just has basic features such as keyboard only to start off with.

Kind Regards,

calebcoverdale, 2019-09-06:
I'm excited to see some updates on the Pi 4 version. 

My boss wants me to order a Raritan Dominion, but I would love to make this little project.  

Do you have a recommendation on the hardware that would be needed for the Pi 4 version so I can get those ordered in ahead of time?

SterlingButters, 2019-09-06:
The goal of our project is require only the Pi4 and a video capture device. There are some oddities with the CSI video capture on the Pi4 so we have chosen to go with a USB capture for now (most should work). 

You can track my progress here but once it is working I may close source: https://github.com/SterlingButters/ip-kvm-interface/tree/dev?files=1

Currently I am able to capture keyboard and mouse input. All that's left if parsing the captured data to send to the target PC (assuming no issues with video capture). From there it'll like just be some fine tuning that's needed.

jtoland, 2019-09-07:
The issue I see with the pi 4 is the power consumption. 

This would be ok for computer with USB 3 or type c but this would not work well with servers as the output is mainly USB 2.

That's why I have chosen to go with the zero.

I have had a look at the link you provided and will have to test it at some point.

Thanks for sharing.

mdevaev, 2019-09-11:
@jtoland @SterlingButters @calebcoverdale 
I have a very similar project in almost production-ready stage (but without documentation lol). My implementation is able to capture images using HDMI and DSI-2, switching video modes, hardware coding on GPU and is capable of outputting more than 30 fps. Right now I can use any version of pi (started from 2) and one arduino with atmega32u4. I have a very advanced backend in which you can implement plugins to support OTG. For ease of development, I created a toolkit that is able to assemble and install the required configuration of the operating system with just one command.  Perhaps we could combine our efforts?
[image]
[image]

SterlingButters, 2019-09-11:
@mdevaev Nice project! Definitely not opposed to combining efforts however I'm fairly confident that my code will not mesh well with yours. The intent of my project structure focuses on everything being very lightweight - using reliable existing libraries and features of node/js. Additionally, even though the project will support the Pi4 (maybe Pi0 because also supports OTG), it will not require any additional hardware besides maybe a USB capture card (working on B101 HDMI CSI2 bridge). Because the project is so lightweight, installation and documentation will be exceptionally easy. Unfortunately I've just been without wifi for several days so I haven't been able to put forth the progress I'd like to. Maybe you could detail some of the advantages of your project - I notice in your picture something about mass storage? Obviously no one wants to abandon their own project but if you are familiar with node/js I'd be happy to bring you aboard.

mdevaev, 2019-09-11:
@SterlingButters 
In fact, my project does not require specific hardware. Everything that you saw in the photographs is optional, and can either be turned off or reconfigured in such a way as not to use additional equipment except for a video capture device. I did a lot of work so that any possible hardware configuration could be described with a simple file, and the user could use standardized APIs, interfaces and much more, regardless of how the KVM works inside.

A few features that I already have:
- Microservice architecture - the system consists of a some separated parts that perform a strictly defined task.
- [Backend](https://github.com/pikvm/kvmd) with clear API that can be used for scripts and alternative UI (for example, you can make a desktop application);
- Own [MJPG streamer](https://github.com/pikvm/ustreamer) written on C than can change the resolution in real time, report statistics about the video and much more (see details in README).
- Ability to use any video capture device.
- Extensible authorization methods - you can configure multiple KVMs so that they use a common password database.
- IPMI BMC support (you can use ipmitool for power management in your existing network infrastructure).
- Mass-storage device based on flash drive (but! can implement OTG for backend).
- A ready-to-use operating system that can be assembled using `make build` and installed to flash using `make install`.
- All components in packages have their own repository. In addition, I developed a special tool that allows you to build the operating system as if it were a regular docker container. It is very convenient and makes debugging easier.

Languages: C, Python, pure JS for Web UI.

jtoland, 2019-09-11:
@mdevaev 
Send me a link to your one-line setup and i will have a play and let you know.
Also what is the git hub page

Spamm00r, 2019-09-16:
Hi guys,

I'm very interested in your projects. And I'm also looking for a low cost generic IP KVM solution.

But I have issues with Raspberry and I prefer to go another route. 

The problem with raspberry is that it is very selective for the hardware. I have 2 capture devices that I can not use because there are no drivers available for raspberry. Also using raspberry requires linux skills. And I as many other users are only able to copy and paste commands others have put in their tutorials. If something isn't working, I'm stuck. Like the missing drivers for my capture cards. Someone with skill may be able to create the drivers, but unfortunately I'm a Linux dummy.

If I now start buying all the right hardware for raspberry that is plug and play compatible, i'll end up with a higher bill than just buying a server Mainboard with vPro or IPMI. ALso right now with all the hardware required for the solutions posted, the Bill is somewhere above 100$.

Also the dozens of cables required to make the connection is problematic and error prone.

I wanted to cut costs and use an old vPro system instead of a raspberry. Old system having vPro capability can be bought for less than 30$, which is cheaper than a rapsberry. The advantage is that it is more powerful than a raspberry, and is managable out of band and you can run any OS you want on it, and if needed boot it bare metal. Thats perfect.

Right now my HTPC Raspberry system keeps freezing every other day and needs a manual hardreset  by pulling the plug. VPro system however are rock solid and reliable. But of course only very few systems have vpro capabilities.

Of course the ultimate Aim is to control another system out of band that has no out of band capability. In my case I wanna be able to control AMD systems or non vPro Intel systems. For example non Q Chipset systems and be completely flexible about the system I use and not stuck to Q-Chip-based Boards and CPU's.

I have already several cheap vpro capable system, that I can VNC into on Bios level.

I only need Video Input and remote keyboard functionality to control any other pc.

I can handle Video Input in VLC on windows, but I have trouble controlling other system keyboard. 

I have looked at several options I think that the Pi0 method you are employing is the best solution.

I have read that using an Arduino is cheaper, but I have trouble finding proper documentation about it.

Can you guys advice me on porting your Keyboard control method to windows?

I'm sure its pretty easy to connect the Pi0 to a PC via USB or Serial Port and use it as a Keystroke Passthrough solution.

What hardware do i need to connect to the Pi0? 

I'd prefer to use a Pi0 WH and SSH into it wirelessly and get Keystroke pass through capability via ssh window without any wired connection between my vpro system and the Pi0. Is that possible?

It would be awesome if someone could describe the steps for that.

mdevaev, 2019-09-16:
@jtoland https://github.com/pikvm/docs
If you have any problems or questions, you can contact us using Discord: https://discord.gg/bpmXfz5

mdevaev, 2019-09-16:
@Spamm00r You must understand that any IP-KVM, as far as it would not be advanced, can never be more functional than the IPMI built into the motherboard since it has more access to the low-level functions of the hardware. If you need the cheapest IP-KVM, you can buy a compatible video capture device (from any of our projects) and use it.

> I'd prefer to use a Pi0 WH and SSH into it wirelessly and get Keystroke pass through capability via ssh window without any wired connection between my vpro system and the Pi0. Is that possible?

I understand correctly that you want to have just a wireless IP-KVM, consisting of a minimum number of parts?

Spamm00r, 2019-09-16:
@mdevaev Yes. All I need is a reliable method to pass through keystrokes with minimal hardware requirements and preferably wireless to the Target machine at BIOS level.  

I was thinking that it should be possible to attach a Pi0 to the USB port of a Target and then SSH into that wireless Pi0 to issue Keystrokes to the target. 

There are "Bad USB" devices like a: Wireless HID CJMCU-3212 that may achieve this.

I would however prefer a Pi0 as it is more versatile and this project here has already achieved this with a Pi0.

I understand that i will be missing advanced features such as USB/PCI redirection etc. But that*s OK for me. All I want is a simple and cheap possibility to issue keyboard commands at BIOS level to the target. 

For resetting the target, i can use a App controlled power socket to hardreset the target if the need arises. So clean and simple wireless solution. Just plug the power cord in the smart power socket and you can start and reset the Target pc. 

I know how I would obtain the Video from the target with a capture device. Thats the easy part. However the Keystrokes over SSH to the Target is the difficult part. I need help with that.

The project you are working on looks awesome. But it looks overkill for my purposes. Also Your hardware list is quite long. may I ask what the hardware costs would be for the setup you are using?

mdevaev, 2019-09-16:
@Spamm00r 
> May I ask what the hardware costs would be for the setup you are using?
~$100 or less. We plan to make a minimum implementation on Zero Pi. It will not be too fast, but probably in a couple of months we will be able to provide something.

Spamm00r, 2019-09-16:
OK. That sounds cool.  I think the most expensive part is the HD capture device.

mdevaev, 2019-09-16:
@Spamm00r 

If you want to use an analog video capture device, you can keep it at $100. When using an [HDMI bridge](https://auvidea.eu/b101-hdmi-to-csi-2-bridge-15-pin-fpc/), the price will be approximately $ 200, since this part is expensive in itself, but the image quality will be incomparably higher. Using my own software, I was able to watch a movie in HD quality.

jtoland, 2019-09-16:
Hi I have found the csi to hdmi a lot cheaper and If we can get the pi zero hid working fine then I think we can get it done for less than 40 per device 

￡25.65  35％ Off | Lusya Raspberry Pi HDMI to CSI-2 Adapter Board HDMI input up to 1080p25fp G4-006
https://s.click.aliexpress.com/e/t05vBNiC

I have one of them and it seems to work fine for me on the pi zero

I am just waiting for more stuff to arrive to see if I can also get vga working by converting it.

mdevaev, 2019-09-16:
Nice. I'll buy myself one for the test. But if it's designed exactly the same as B101, then it will work perfectly with my streamer.

calebcoverdale, 2019-09-16:
I just bought one as well! Hoping it works out.

mdevaev, 2019-09-24:
Just tested this configuration
- RPi4;
- Auvidea B101 HDMI-CSI2 bridge;
- External flash drive to storing ISO images;
- FT232 (only to get serial console & power, it's not necessary for real usage).

RPi4 has OTG in USB-C, it's perfectly emulate CD-ROM and keyboard. To prevent KVM from losing power during server shutdown, power must be supplied through the GPIO pins. I think that we need a way to turn off the power through USB-C, so as not to overload the port. I managed to get video capture to work with my streamer and enable OTG. A separate flash drive is needed to avoid wear of the SD card with the root file system when overwriting images and the risk of power failure without unmounting.

It looks like soon I will adapt the Pi-KVM to work with this configuration. I think this is better than using Zero W because RPi4 has Ethernet and USB host.

[image]

calebcoverdale, 2019-09-24:
I’m going through buying these parts now! 

Hoping to emulate what you have. Could you make a dump of your SD card once you are done? Or do I just follow the instructions to get it setup

mdevaev, 2019-09-24:
@calebcoverdale I will update the instruction and you can simply repeat the necessary steps: https://github.com/pikvm/docs

mdevaev, 2019-09-24:
@calebcoverdale There are some difficulties in providing a ready-made image of the system due to the fact that it's read-only. It means that we need a some way to generate ssh keys and other things. Maybe I'll do it in the future but now I have other things to do in this project.

mdevaev, 2019-09-30:
I've got a mouse, keyboard, and mass storage using OTG on RPi4. It takes a little debugging, but it works pretty well.
[image]

mdevaev, 2019-09-30:
Okay, just received a package in the mail with Chinese CSI-2/HDMI bridge: https://aliexpress.com/item/4000102166176.html

It works perfectly and without complaints, so yes, now we have a cheap HD capture device.

bobbintb, 2019-10-01:
I'd be very interested in getting this to work with a Pi Zero. I've already got a lot of older Pis laying around and I'd rather not get another one if I can just use a Zero. Looking forward to updates.

mdevaev, 2019-10-01:
Basic functionality will work, but don't expect much from Zero: it has a weak CPU. Encoding HD video requires a lot of resources.

In a few days, my ZeroW will come to the post office and I'll tell you how things are with it.

bobbintb, 2019-10-01:
I'm excited to hear how it goes. I expect it to be just fine for my purposes but you never know. I don't need high resolution. It sounds like the CSI2/HDMI bridge and a Zero W would be all the hardware I need, except for cables and chargers and such. That sound about right?

mdevaev, 2019-10-01:
Yep.

Spamm00r, 2019-10-01:
Its good to have a cheap HDMI input. I experimented with analog input and the quality is unusable.

Unfortunately, a RPI 4 and the HDMI Input alone raise the price to about 100$. At that price level one can consider investing that money into a KVM Mainboard directly. 
I was hoping to have a low cost option. With a DIY KVM I was hoping to add KVM functionality to cheaper hardware. But 100$ is right on the edge of what a DIY KVM can cost for it to make sense financially and technically.

As of now I will probably not buy another RPI and instead opt for one of those tiny/micro pc's that are able to run Windows 10 on T CPU's. You can have those for under 100$. A 70$ RPI4 is just too expensive compared to these offers. And a RPI is limited in its use cases compared to a fully fledged tiny PC.

Appelg, 2019-10-06:
@mdevaev I want to use this on an Intel Nuc. It only has USB ports. Is that possible? I.e I want mouse/keyboard commands to go over USB, is that doable?

mdevaev, 2019-10-06:
@Appelg do you mean that Intel Noc acts as a KVM or you want to use RPi to control Intel Noc?

Appelg, 2019-10-06:
@mdevaev I want to control the NUC with the RPi KVM.

mdevaev, 2019-10-06:
@Appelg yes, it's possible. Currently I have two hw implementations:
1) Arduino keyboard/mouse emulator that connects to GPIO on RPi and USB on the host. This is the only option if you use RPi 1, 2 or 3.
2) OTG emulator based on RPi 4, it will also be available on Zero W. It also just connects to the host, the only difference is that you don't use additional hardware and get the opportunity to use a CD-ROM emulator.

Appelg, 2019-10-06:
@mdevaev thats awesome, I’ll try it! Thanks a lot!

mdevaev, 2019-10-06:
@Appelg you're welcome! Join to our discord: https://discord.gg/bpmXfz5. Developers and users are available in the chat, you can quickly get an answer to any question.

bobbintb, 2019-10-07:
> Its good to have a cheap HDMI input. I experimented with analog input and the quality is unusable.
>
> Unfortunately, a RPI 4 and the HDMI Input alone raise the price to about 100$. At that price level [...]

I think $100 is a pretty high estimate. For one, the Raspberry Pi 4 is $35, not $70. Even the higher memory ones aren't that much. I suppose if you include a bunch of the peripherals it could add to the cost but a lot of hobbyists have extras of those. The highest model of Pi 4 is rather overkill for this project anyway, although if you're like me I'd probably get it anyway if I need to repurpose it for a future project that requires more resources.

I'm interested in trying it with a Zero W. I know it's weak, but I don't think it would need to be very powerful for this. I except the HDMI bridge will get cheaper or eventually other Chinese manufacturers will start undercutting each other. I'm thinking this can be done for $50.

Spamm00r, 2019-10-07:
Where did you find a RPI 4 for 35$? 

I can only find RPI 4 2GB for 50 and the 4 GB version for 60 excluding shipping. And with only 10 $ difference, I would opt for the 4Gb version just in case. Add 30 for the HDMI input and you are at 90. Not counted in are sd cards, Power Input and USB cables etc. 

I bought yesterday a used 4th gen vPro system for 30$. That has out of Band IP KVM on board.

That's why I'm hesitant to buy another Raspberry. With all those additional things that you need to get your raspberry pi up and running, at least for me, it makes more sense to just buy a used PC system which is far more powerful.

I originally bought my first PI for KODI as a HTPC. But after having to buy two wifi dongles, because the first one wasn't compatible, a Blue Tooth dongle, my thrid SD card, because two SD's died inbetween and another Power supply, becouse the first on wasn't strong enough, I ended up with close to 100 $.  Not counting the countless hours I wasted researching, setting up and configuring the pi. 

I'm not so convinced about Raspberry anymore. On top of that, I'm more often updating and patching outdated Software on the raspberry than using it.

For example I use the raspberry  as a Nvidia gamestream receiver with moonlight. I have to update the software nearly everytime I want to use it and it will regularly crash after a few minutes. On windows, "It just Works!"

Also my 24/7 RPI that I use to SSH into, regularly is not connectable, even though I have a watchdog installed.

I'd rather stick to windows.

I'm hoping to get this project working on the PiZero, which can be bought for 16$. I don't really wanna buy another Pi for 60$. 

//Sorry for the rant!

mdevaev, 2019-10-07:
@Spamm00r This will work for Zero. I just need to write the correct configuration for this board. I'm waiting for a flat cable for HDMI bridge.

However, I should note that the price of a home-made KVM does not exceed $120 even taking into account the cost of a new RPi4. It’s much cheaper than commercial KVM ($250 minimum). Moreover, the home-made KVM already has much greater capabilities.

Spamm00r, 2019-10-07:
@mdevaev I'm looking forward to the RPI zero implementation.

I agree that the diy KVM is far cheaper than the solution offered by the commercial IP KVM providers. My issue is that 120$ price tag for the DIY solution is close to the edge of where it makes from a price perspective not sense to invest in a DIY solution.

For example I'm thinking to switch from my current Intel vPro setup that has a IP KVM on board to a AMD ryzen system in the future, which is about 100 to 120$ cheaper than a Intel system. But the Ryzen systems has no on board IP KVM solution at all. Now adding the costs of a DIY IP KVM eradicates the price advantage of the current AMD systems.

That way one can just save the hussle and stick with intel's vPro solution.

That's why I'm hoping for a cheaper PI zero solution.

bobbintb, 2019-10-07:
@Spamm00r Raspberry Pi base model has always been $35. I haven't actually bought a 4 yet. All the extras can add to the price but a lot of people, like myself, have extra SD cards, chargers, cables, etc. Like you, even though the higher model is overkill for this project, I'd probably still get it because the price difference isn't much and I might end up repurposing it.

mdevaev, 2019-10-08:
@bobbintb RPi4 1Gb costs minimum $55.

bobbintb, 2019-10-08:
> @bobbintb RPi4 1Gb costs minimum $55.

That's for the 4gb model.
1gb - $35
2gb - $45
4gb - $55

Unless you are getting them at a markup. It's always been $35 from authorized sellers.

mdevaev, 2019-10-08:
Could you give a link to a store with such prices? The cheapest I found was worth 47 bucks:
https://www.amazon.com/Raspberry-Model-2019-Quad-Bluetooth/dp/B07TD43PDZ

bobbintb, 2019-10-08:
> Could you give a link to a store with such prices? The cheapest I found was worth 47 bucks:
> https://www.amazon.com/Raspberry-Model-2019-Quad-Bluetooth/dp/B07TD43PDZ

I edited my reply with links. These links are all directly from raspberrypi.org.

ZafotheNinja, 2019-11-10:
Hi all, wanted to ask if there is any update on the pi zero version as I have a computer that could use it (no video capture required)

Also wanted to ask about the possibility of OTG video capture, is it possible to emulate a usb monitor?
https://www.instructables.com/id/Raspberry-pi-Zero-as-USB-to-HDMI/

jtoland, 2019-11-10:
@ZafotheNinja Pi Zero version is currently still in testing and should be available soon.

I have had a look at OTG capture and this is not possible from what I have read you would need some kind of USB graphics converter. if you are using a computer without a graphics card I would recommend having a look at some kind of USB video card such as
https://www.amazon.co.uk/Adapter-Monitor-Adaptor-Converter-Compatible-Black/dp/B075K86J5R 

With that, you should be able to plug it into the CSI to HDMI bridge and get what I believe you are looking for with some kind of output.

You can use the software as non-video only but I believe this defeats the reason for the project as you want to do stuff remotely and will need some kind of screen to be able to see what you are doing

I also believe you should join the discord: https://discord.gg/bpmXfz5 as if you have any questions they can be answered quicker as the developers are really handy to speak to and bounce ideas off.

ZafotheNinja, 2019-11-10:
Currently I just need a way to press the power button remotely and monitor the power led. So input/output besides that is not required.

Mostly for a hard reset or to power on if accidentally turned off, I have I vnc/remote desktop that starts up when windows is fully booted, currently no need for bios access, but that would be nice in the future.

Spamm00r, 2019-11-14:
I may add that remote BIOS Level control is also possible via Serial over LAN. 

NON-GUI BIOS's can be controlled via a serial connection by enabling console redirection. That way you can control an out of band pc with just a serial connection. No Capture Card required. And once windows or Linux is loaded, you can remote access it via rdp or vnc.

mdevaev, 2019-12-04:
Okay, I've finished implementing KVM for ZeroW and HDMI-to-CSI2 bridge. It supports mouse and mass storage emulation (CD-ROM or Flash at your choice). In addition, I added support for Wake-on-LAN. I'll be glad if someone tries it: https://github.com/pikvm/docs
[image]
[image]
[image]

Neo2199, 2019-12-30:
Nice work. I am going to build one on pi4. Would be nice to see performance comparison between the zero and pi4. I know you mentioned that performance on zero is super slow, but can you put some numbers? how many fps?
I think most of the people are building this remote KVM just to do computer reinstall or some basic diagnostics. So not sure how much performance is needed in order to do few simple task.

mdevaev, 2019-12-30:
@Neo2199 Right now I'm on vacation away from my projects, so I can't answer your question yet. The difference is about 15 vs 25 fps as I remember. But the most important thing is that it is almost impossible to upload an iso image to zero at the same time as the streaming. It takes up all the resources.

Neo2199, 2019-12-30:
Sure, I think this gives me much better idea on what is happening. Out of curiosity, can you set some priorities? So let say the streaming has always access to about 50% of CPU?
What mainly interested me on zero w was small footprint, low price, and low power consumption so it can be powered from single USB if wanted. 
Anyhow, I think this is a great project as is. Thanks.
I have just ordered parts, so will be some time till I have a working prototype.

mdevaev, 2019-12-30:
@Neo2199 
> 50% of CPU?
Yes, but it's not just the CPU, the I/O capabilities of Zero also leave much to be desired.
Write to discord if you have any questions.