Fmstrat/diy-ipmi #44: Diagram has incorrect wire to relay

grantcurell, 2021-03-12:
No idea if this repo is still active but it's a cool project and I bought the parts and am building out a newer version with a current version of Raspbian. I did use the suggested relay and found that to get the power to work it was actually slots one and two on the relay:

[image]

SterlingButters, 2021-03-12:
@grantcurell Check out: https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwj8g6qd96vvAhVIbs0KHQ7ICesQFjAAegQIAhAE&url=https%3A%2F%2Fgithub.com%2Fpikvm&usg=AOvVaw3OLdyXB-URtDszGfA8VTVo

grantcurell, 2021-03-12:
> @grantcurell Check out: https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwj8g6qd96vvAhVIbs0KHQ7ICesQFjAAegQIAhAE&url=https%3A%2F%2Fgithub.com%2Fpikvm&usg=AOvVaw3OLdyXB-URtDszGfA8VTVo

Thanks man! I really appreciate it! I'll do that.

Fmstrat, 2021-03-12:
While this project still works and has users, and I do maintain it somewhat, my understanding is it did become the kickstart for a number of others that are more modern (Pi-KVM) being one of them.

Granted, I'm curious about the wiring issue you mention, as that's how I have mine wired up. I'll have to do a comparison.

grantcurell, 2021-03-13:
@Fmstrat - I will in no way be hurt if you tell me I did something silly 😂

https://youtu.be/IByzL-0YYdg